prusa3d/PrusaSlicer #11470: Ability to use macros in Custom G-Code at layer

Waillant, 2023-10-15:
**Is your feature request related to a problem? Please describe.**
When I add custom g-code at layer (for example to pause a print to insert a nuts), I need to park nozzle, reduce temperature and then restore the temperature to the original one

**Describe the solution you'd like**
Ability to use prusa macro in custom g-code at layer (on orange +)

**Describe how it would work**
I need to use macros like:
M104 S{temperature[0] * 2/3}

**Describe alternatives you've considered**
Actually I need to set a fixed temperature like:
M104 S150
but 150 degree may be correct with PLA but not with ABS

I use PrusaSlicer 2.5.2 on gentoo linux

m-cas, 2023-10-18:
Did you try with slicer v2.6 (and above)? From 2.6 there is a lot more support around macros...

Waillant, 2023-10-18:
Actually 2.6 Version is not available on gentoo and in the documentation
seems that this function is unavailable.

If you say that this function can exists on new Version, maybe I can try to
install it manually.
Thank you.
Bye, Nicola

Waillant, 2023-10-19:
Hi everyone, I have installed version 2.6.0 in Gentoo and I confirm that it is not possible to use macros when inserting custom gcode into a layer.
It would be a very convenient feature...

Thanks.
Bye Waillant.

closed 2026-09-29 by prusaslicer-bot[bot]