prusa3d/PrusaSlicer #11472: Prusa Slicer - Arachne leaves unfilled gaps at junction interfaces between perimeters and top

Zama747, 2023-10-15:
### Description of the bug

The Prusa Slicer with arachne activated does not fill the gaps that are in the transition interface between top layers and perimeters.

I understand that with the arachne activated it is not necessary to fill gaps.
However, in conditions where the gap is left due to the lack of a top layer (less than the minimum extrusion size), the surrounding perimeters should be wider to fill the gap left.
The arachne only starts to work if there is a gap large enough to be filled with a top layer, it does not work in intersections between small gaps in top and perimeter layers.

The problem with this specific part can be solved by adding 2 perimeters to the printout. However, if I increase the scale of the part, the problem will repeat itself when there is a very small gap between the perimeters and the top layers.

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0

### Operating system

windows 10

### Printer model

Ender 3

FrHePo, 2023-11-14:
Same with me (PrusaSlicer 2.7.0-beta1). Arachne leaves open gaps (inside hole) where the classic perimeter generator fills the gaps.

[3mf+gcode.zip](https://github.com/prusa3d/PrusaSlicer/files/13343680/3mf%2Bgcode.zip)
[image]
[image]

antimix, 2024-10-06:
I have the same issue with Aracne on many prints, and for this reason a slice almost 90% with Classic.
It would be nice to have this bug fixed.

closed 2026-09-29 by prusaslicer-bot[bot]