prusa3d/PrusaSlicer #11475: Slicing multiple beds at once after "Arrange". Save multiple files or send multiple via Prusa Link

Valthol, 2023-10-16:
**Is your feature request related to a problem? Please describe.**
For me it is always ongoing scenario and I do believe that for more of us. I pust all of the objects in PrusaSlicer that I have on order to print, I arrange them and I get 3-5-10 arranged beds ready to be sliced. I want them all to be sliced, this is why I have put more files that I know that fits in single bed. I have to slice first, save, select all objects, move them by given distance to the left that I have calculated, slice, save, select all, move by distance, slice, save. Pretty time consuming and easy to automate and make simpler.

**Describe the solution you'd like**
The arrange function already arranges all of the objects in perfect bed sized rectangles. Only first rectangle is sliceable, the rest need to be moved and sliced. It would be very time saving if all of those prepared beds could be sliced at the same time or, knowing it is resource consuming, one after another but without the user interference.

**Describe how it would work**
Based on what user needs - do I need to save all of those beds as a gcode file in my local drive, or send them automatically to printers via Prusa Link it could:

a) Get a part of the file name from us in the options hard coded by us (we would not need to input the part of name each time) and save each sliced bed with given name + printing time? material cost? the user could choose what would be added to the file name
b) pop up a box with the printers from our Printer Profiles with checkbox to choose which of them to use and then automatically send each bed to next printer that we chose with checkboxes via PrusaLink like we normally do using the "Send to printer" button ned to Export G-code.

This whole option that I am talking about had to be checked in the Advanced menu. The program could be creating additional virtual visualisation of beds for each arranged full bed for us just to make sure that every part will fit nicely in the beds (currently all of prepared beds, exluding first one, are without any graphic representations of the bed, floating in the air).

**Describe alternatives you've considered**
For sure the option with saving g-code files only locally will be easier to implement for starter and it would be already a very good boost for the print farms productivity, but the second option to integrate it with massively uploading files via PrusaLink would be a gamechanger.

**Additional context**
Just a very good speed up :) If you upload more files than the bed can fit and you are managing any kind of a print farm (even for hobby use) - for sure you are going to slice all of those beds anyway.

sarusani, 2023-10-17:
Similar requests: #9324, #8575, #8317

kubispe1, 2023-12-22:
Hi, thanks for your ideas (in in shorter summary):
- Save all of the sliced beds as G-code files to the user's local drive.
- Automatically send the sliced beds to the user's printers via PrusaLink. 
I would also extend it to PrusaConnect which enables create a print query
Let's see what we can obtain.
There is a thread I would like to use to summarize ideas around "Multiple Build plates" features #10704

closed 2026-09-29 by prusaslicer-bot[bot]