prusa3d/PrusaSlicer #11478: raccord en mode vase

18pool49, 2023-10-17:
### Description of the bug

bien que le tranchage est demandé en mode vase il y a trace , 
peut visible a l'écran >>>>
[image]
mais parfaitement visible sur le lit 
[image]
j'ai vérifié le Gcode a l'écran de Prusa , il y a bien un repositionnement a chaque cycle (tour ) 
[image]

[image]
[image]
[image]
j'ai fait la même pièce avec Cura et Ideamaker en mode vase aussi mais il y a pas de trace visible c'est vraiment du mode vase 
[image]
a droite c'est Prusa et a gauche c'est Ideamaker Cura est encours de dépose et je ne distingue aucune anomalie 

### Project file & How to reproduce

[ampr tete mort 90° projet raccord vase.zip](https://github.com/prusa3d/PrusaSlicer/files/12961303/ampr.tete.mort.90.projet.raccord.vase.zip)

réalisé en hauteur variable et vitesse variable des surplombs
qui ne sont pas la cause >> après essais <<

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.1

### Operating system

W11

### Printer model

geeetech A10M

neophyl, 2023-10-18:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/2841

FidelCapo, 2023-10-23:
As @neophyl wrote, duplicate of https://github.com/prusa3d/PrusaSlicer/issues/2841
Closing.

closed 2023-10-23 by FidelCapo

18pool49, 2023-10-23:
résultat des courses ; il me faut donc attendre la saint "Glinglin" pour disposer d'un mode VASE ou plutôt , un mode "spiroïdal" ,pour ne plus avoir ce raccord >> disgracieux  <<
question idiote >> comment fait donc la concurrence << ils sont plus malin  ! ! ! !