prusa3d/PrusaSlicer #11480: Bug with Arrange with multicolor prints on 2.6.1

LivingSynthesis, 2023-10-18:
### Description of the bug

When you arrange a multicolor print in 2.6.1 it will be removed from the print area.

When I click arrange in 2.6.1
[image]

When I click arrange in 2.6.0
[image]

It's probably an unexpected side effect of the new arrange features somehow conflicting with the wipe tower, totally understandable! Or perhaps there's some new setting I'm missing that corrects this that I was unable to figure out? Definitely let me know if that's the case! I'll just stick with 2.6.0 until a fix is available.

### Project file & How to reproduce

1) Add a mutlimaterial model
2) Click arrange
[arrange bug.zip](https://github.com/prusa3d/PrusaSlicer/files/13034888/arrange.bug.zip)

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.1

### Operating system

Win 11

### Printer model

Prusa i3 mk3s+

LivingSynthesis, 2023-12-18:
This issue is still present on 2.7.0 can someone please help?

dwineman, 2024-06-14:
Still present and very annoying in 2.7.4.

[image]

dwineman, 2024-06-14:
@LivingSynthesis a better title that may get more notice would be something like "Auto-Arrange moves entire model off of build plate." Describe the problem, don't just say "bug with". Please edit your title so this gets attention. It's a serious usability issue that's been ignored for eight months.

LivingSynthesis, 2024-06-14:
Great tip, thanks!

ericbjones, 2024-07-10:
This describes the same issue: https://github.com/prusa3d/PrusaSlicer/issues/10859

dwineman, 2026-09-18:
This issue is still present in 3.0.0-alpha11. When auto-arranging bed 1 with a large multicolored object and wipe tower, the object is moved to bed 2. Arranging bed 2 then moves the object back to bed 1.