prusa3d/PrusaSlicer #11482: 'Merge' doesn't work after cutting-with-connectors

puterboy, 2023-10-19:
### Description of the bug

If you cut-with-connectors, you lose the option to right-click 'merge' either of the cut objects with any other object - whether that object was created/imported before the cut action or after.
If you cut-without-connectors, then the right-click 'merge' option works normally.

### Project file & How to reproduce

If I cut *without* adding connectors, there is a 'Merge' option with a subsequently added shape

1. Import STL
2. Cut STL *WITHOUT* adding connectors
3. Add shape
4. Select one of the cut objecs and the added shape
5. Right Clicke -> Merge option exists

If I cut *with* connectors, there is *no* merge option after adding new shape
1. Import STL
2. Cut STL *WITH* adding connectors
3. Add shape
4. Select one of the cut objects and the new shape
5. Right Click -> NO MERGE OPTION

Similarly, if I add the shape before cutting-with-connectors...

This seems like a bug to me...

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.1

### Operating system

Windows 11

### Printer model

Prusa MK3S+

YuSanka, 2023-10-20:
Thanks for your report. But it's intended behavior. We don't allow to merge, separately scale or reload from disk actions for objects cut with connectors. Because after that we can't guaranty objects stability.
But if you are sure, that you need to perform some of forbidden action, you have to select "Invalidate cut information" in object context menu.

closed 2023-10-20 by YuSanka

puterboy, 2023-10-20:
Thanks for the explanation.
I can understand why separate scaling will more often than not mess up connectors but that is not necessarily true with merging -- so long as the merged object doesn't intersect with the other cut object -- and regarless why would you assume that a user is not smart enough to avoid such problems?

Moreover, the "dangers" you raise apply equally whether the cut has connectors or not. i.e., if you separately scale cut objects or merge something that intersects with the other side of the cut, there will likely be a problem regardless of whether there are connectors or not.

Finally, at a minimum, this restricting behavior and the ability to override it should be clearly documented. It is not at all obvious that selectiong "invalidate cut information" restores the ability to merge or scale.

Better yet, why not allow the behavior but issue a warning that you are potentially messing up the connector aligment. This is much less heavy-handed and also more self-explanatory than just blindly shutting off the functionality. Also, I may want to selectively and intentionally merge or scale without "invalidating (all) cut information".

I respectfully suggest that you rethink what is the most user-friendly workflow here.