prusa3d/PrusaSlicer #11483: Assertion '!this->empty()' failed with Arachne

fabienInizan, 2023-10-19:
### Description of the bug

The slicer is crashing due to an Assertion Error when using the Arachne generator, during infill generation according to the progress bar. The slicing completes with no error with the Classic generator.

The full error:
```
/usr/include/c++/13.2.1/bits/stl_vector.h:1220: std::vector<_Tp, _Alloc>::const_reference std::vector<_Tp, _Alloc>::front() const [with _Tp = Slic3r::ExtrusionEntity*; _Alloc = std::allocator<Slic3r::ExtrusionEntity*>; const_reference = Slic3r::ExtrusionEntity* const&]: Assertion '!this->empty()' failed.
Abandon (core dumped)
```

This is version 2.6.1 running on ArchLinux.

### Project file & How to reproduce

Here is the [3MF project](https://github.com/prusa3d/PrusaSlicer/files/13046720/MarbleMachine3.zip) to reproduce the issue.

This is not targeting a Prusa printer but a RatRig V-Minion. I couldn't reproduce when slicing for MK4 or a Mini.
It also only happens with this specific configuration, using a modified 0.15mm quality profile. Other profiles don't exhibit the issue.

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.1

### Operating system

ArchLinux

### Printer model

RatRig V-Minion

neophyl, 2023-10-19:
Happens on windows 10 too.  Crashes on infill.  It seems to be the bridge flow ratio.  If I change the 0.7 to something else like 1.0 it slices fine.  Even changing it to 0.71 allows it to slice for me.

rydrman, 2023-12-03:
I'm on Fedora and am seeing this with either generator and with thick bridges off as well. I am also slicing for both mk3 and mini, but using a custom config that could have some other setting that is causing the same error

closed 2026-09-29 by prusaslicer-bot[bot]