prusa3d/PrusaSlicer #11484: Split to Objects crashes

pylorak, 2023-10-20:
### Description of the bug

On specific model, split to object crashes PrusaSlicer 2.6.1.

### Project file & How to reproduce

Steps to reproduce:

1. Open attached project [V6_Plenum[2020_extrusion].zip](https://github.com/prusa3d/PrusaSlicer/files/13054439/V6_Plenum.2020_extrusion.zip)
2. Left click on 3D model to select it (if not already selected)
3. Right click on 3D model to open its context menu
4. In the context menu, select Split -> To objects
5. Wait. 
Object is seemingly split, but PrusaSlicer crashes after 2-3 seconds without any further actions

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.1+win64

### Operating system

Windows 11 (Build 22621.2428)

### Printer model

Voron Trident

PaparazziN, 2023-10-21:
Hi @pylorak!
I tried replicating this with the PrusaSlicer-2.7.0-Alpha1 and it does indeed crash for me there as well. I did however find something interesting when splitting to parts, as it splits into five different parts. However only three of these (the big part and two clips) seem to have volume, when deleting these three parts PrusaSlicer crashes. On the surface it seems similar to #8931 but that issue should be fixed. 
A quick fix I found was running the "Fix by Windows repair algorithm" and then splitting to object, however that doesn't really solve the underlying problem.

pylorak, 2023-10-21:
Hi @PaparazziN !

The model comes unmodified from the Nevermore V6 air filter project found here: https://github.com/nevermore3d/Nevermore_Micro/blob/master/V6/STL/V6_Plenum%5B2020_extrusion%5D.stl

I don't know what those other volumes are. Expected are three parts after the split, the big part and two clips, as you wrote.

Issue #8931 is already fixed and closed, but it definitely seems similar. 

The workaround with repairing inside PS is nice as it allows me to work with this model inside PS for now until this issue is fixed, Thanks for the tip.

lukasmatena, 2023-10-23:
Thanks, reproduced. We will look into it. It is not the same as #8931.
SPE-1983

lukasmatena, 2023-11-11:
Fixed in [2.7.0-beta1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.0-beta1). Closing.

closed 2023-11-11 by lukasmatena