prusa3d/PrusaSlicer #11485: Rectilinear Infill Pattern causes app to crash on specific model

seemly, 2023-10-20:
### Description of the bug

I have had multiple crashes in PrusaSlicer for a specific model, but only when attempting to use the Rectilinear Infill Pattern.

The specific model URL:
https://files.printables.com/media/prints/1097/stls/7092_11a48ad8-9bef-46a1-a1d2-b060fc5b1048/parallel-pliers-single.stl

https://www.printables.com/model/1097-print-in-place-parallel-pliers-multi-material-vers

### Project file & How to reproduce

I'm using PrusaSlicer 2.6.1, on Apple Mac mini M2 8GB Ventura 13.4

1. Download the following model:
https://files.printables.com/media/prints/1097/stls/7092_11a48ad8-9bef-46a1-a1d2-b060fc5b1048/parallel-pliers-single.stl
2. Change the Infill > Fill Pattern to Rectilinear.
3. Slice

Voila. The app consistently crashes (for me).

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.6.1+MacOS-arm64

### Operating system

Apple Mac mini M2 8GB Ventura 13.4

### Printer model

Sovol SV06 Plus

JohnOCFII, 2023-10-21:
You'll need to upload a 3MF with your specific configurations, as I didn't have an issue slicing this STL on my Arm Mac with PrusaSlicer 2.6.1 after switching infill to Rectilinear.

seemly, 2023-10-21:
Hi, @JohnOCFII. My apologies for not including it in my opening message.

I'm a novice with 3D printing, so I didn't know what was required.

I have attached a zip file of the 3mf file, as Github is telling me they don't support the .3mf extension.

[parallel-pliers-single.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/13060365/parallel-pliers-single.3mf.zip)

neophyl, 2023-10-21:
Well the good news is that 2.7 alpha 1 doesnt crash with the project :)  Tried it with 2.6.1 on windows 10 and it crashes out every time after getting to infill, but the 2.7 alpha slices to the end.

seemly, 2023-10-21:
> Well the good news is that 2.7 alpha 1 doesnt crash with the project :) Tried it with 2.6.1 on windows 10 and it crashes out every time after getting to infill, but the 2.7 alpha slices to the end.

Ahh, that's good news then!

I thought it was incredibly strange that an infill pattern would consistently make the app crash, but specifically on this model only!

Very odd!

lukasmatena, 2023-10-21:
Thanks. It sounds like a duplicate of #11267 (and others) which was indeed fixed in 2.7.0-alpha1. Closing.

closed 2023-10-21 by lukasmatena