prusa3d/PrusaSlicer #11486: the vase mode body will have a hole if there is a chamfer at starting position

zxy16305, 2023-10-20:
### Description of the bug

When vase mode body starting position is a chamfer, the slice result at this location is lower volumetric speed、 same height、same line width. These factors contribute to the formation of a hole in the printed object at this position.
[image]

### Project file & How to reproduce

[Dustbin-uncut-widerbottom-twisted.zip](https://github.com/prusa3d/PrusaSlicer/files/13057640/Dustbin-uncut-widerbottom-twisted.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.1

### Operating system

windows 10

### Printer model

klp1

lukasmatena, 2023-10-23:
Thanks. The transition is actually by design (298b730589371d39fe4a85851a9c72f87fc22d8e). The extrusion rate of the first spiral vase layer is gradually increased so it is not squished on the lower layer at the beginning (as was hapenning with #3712 and #2795). You are correct that it is not desirable when the first spiral vase layer is an overhang. We may look into it one day, but it is not a 5 minute job.

pedromelim, 2023-12-11:
Encountered the same issue, but im using a 0.8mm nozzle instead of the 0.6mm that @zxy16305 is using.
For a 0.4mm nozzle (and smaller ones) probably is not showing this issue as the bigger ones, since it will cope way better with the low flow rate when doing linear tapering of extrusion rate for the 1st spiral vase.

In theory, the implementation is spot on, but considering a bigger nozzle will not cope well with this, maybe a solution would be to use the layer height limits from the Printer Settings:

[image]

closed 2026-09-29 by prusaslicer-bot[bot]