prusa3d/PrusaSlicer #11487: Cut Connectors Broken when assigning part to other side

ThyOneGuy, 2023-10-20:
### Description of the bug

Cut connectors such as frustum and snap connectors are not present after slicing if you assigned parts to the other side during the cut that made the connectors.

Image of the connectors after finishing the cut
[image]

Image of them missing from the slice
[image]

Picture of the object viewer while the connectors are broken
[image]

If I do not assign any parts to the other side then the connectors appear as normal. Its just really hard to cut up a large complex model without the ability to assign parts to the other side with connectors working.

### Project file & How to reproduce

[Ice Cane.zip](https://github.com/prusa3d/PrusaSlicer/files/13057702/Ice.Cane.zip)

Reproduce by cutting a model through multiple features and assigning some of them to the other side with a right click.
Add connectors such as frustrum or snap
Finish cut and slice
Connectors will be missing from final slice view, and do not print if you try anyways.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.1+win64

### Operating system

Windows 10 22H2 (19045.3570)

### Printer model

Creality CR-10 Smart Pro

lukasmatena, 2023-10-23:
Thanks. Adding reference to #11415. We will look into it.

Jan-Soustruznik, 2023-10-24:
SPE-1979

lukasmatena, 2023-11-11:
Fixed in [2.7.0-beta1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.0-beta1). Closing.

closed 2023-11-11 by lukasmatena