prusa3d/PrusaSlicer #11491: Supports: unsupported bottom perimeters

viesturz, 2023-10-21:
### Description of the bug

The bottom layers of a 90deg overgang area are often parralel to the skin.
The outer edge is supported by the supports, but rest of the area is not, resulting in long unsupported stretches.

The issue is the same with both grid and organic supports.

### Project file & How to reproduce

[Beam-Bottom.zip](https://github.com/prusa3d/PrusaSlicer/files/13060310/Beam-Bottom.zip)
[image]
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.1

### Operating system

Win11

### Printer model

voron 2.4

neophyl, 2023-10-21:
You have Dont support bridges enabled.
Those lighter blue extrusions are bridge extrusion, if you have the option enabled they wont be supported.

Also for an object like this then snug support might be more suitable than organic.

viesturz, 2023-10-21:
Thanks, disabling don't support bridges helped.

There are still two potential improvements here:
- there are cases when bridges are perfectly fine, perhaps a new option can be added to support long bridges every N mm.
- when slicing bridges orient the strokes to minimize bridge length.

closed 2026-09-29 by prusaslicer-bot[bot]