prusa3d/PrusaSlicer #11492: 2.7 alpha 1 Cancel Object RRF Bug.

P-C-R, 2023-10-21:
### Description of the bug

If using the Mark Objects the naming at RRF does not work.

its is

M486 S0
M486 A"Creepy Coffin Box - Single Material.stl"

but should be

M486 S0
M486 S0 A"Creepy Coffin Box - Single Material.stl"

without it RRF is not naming the Objects

see more here
https://docs.duet3d.com/User_manual/Reference/Gcodes#m486-object-cancellation

### Project file & How to reproduce

Just slice and see code

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7 Alpha 1

### Operating system

LINUX

### Printer model

Snakeoil

lukasmatena, 2023-10-24:
Thanks, we will fix it. SPE-1985.

lukasmatena, 2023-11-11:
Fixed in [2.7.0-beta1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.0-beta1). Closing.

closed 2023-11-11 by lukasmatena