prusa3d/PrusaSlicer #11493: 2.7 alpha 1 - Cancel object leaves unnecessary travel moves to canceled object

Gyrxiur, 2023-10-21:
### Description of the bug

When you use cancel object feature, printer still moves every layer to canceled object location and back. I love this feature, but it looks like unnecessary travel moves were forgot to be also canceled. 

### Project file & How to reproduce

Print two or more objects, cancel one of them, printer still unnecessarily moves every layer to it and back.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.0 alpha 1

### Operating system

Windows 11

### Printer model

MK4

SvenPaterson, 2024-04-19:
Not sure if this is the right place to add to this, but I'm on Mk4 6.0.0 fw and Prusaslicer 2.7.1. When I hit cancel object the z axis raises in short bumps along with corresponding retractions. But when it moves to the next object it doesnt appear to re-retract so it starts printing and there is no filament primed to continue appropriately.

lukasmatena, 2024-04-20:
This should be fixed in 2.7.2 and later. Closing.

closed 2024-04-20 by lukasmatena