prusa3d/PrusaSlicer #11495: Avoid crossing perimeters - extremely slow slicing for simple spiral

viesturz, 2023-10-21:
### Description of the bug

Avoid crossing perimeters gets extremely slow, spending >60 sec stuck at 90% for a simple 10 revolution spiral.

What I'm normally using is:
- avoid crossing perimeters enabled
- max detour legth: 100mm

### Project file & How to reproduce

[Body1.zip](https://github.com/prusa3d/PrusaSlicer/files/13060748/Body1.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.1

### Operating system

Win11

### Printer model

voron 2.4

lukasmatena, 2023-10-23:
Thanks, we will investigate.

RyanDMott, 2024-03-22:
This is arguably a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/8414. 

I also have observed a fine-resolution curve causing fine-resolution travel moves (v2.7.2). This not only slows g-code generation but also may bog down the print itself (e.g. when using Octoprint). https://github.com/prusa3d/PrusaSlicer/issues/8414 notes that a coarse setting for g-code resolution does not apply to travel moves. 

I found that reducing the resolution of the model itself, e.g. with the Simplify Model tool, is an adequate workaround in some cases.

closed 2026-09-29 by prusaslicer-bot[bot]