prusa3d/PrusaSlicer #11496: Connectors no showing when cutting

rt6151, 2023-10-21:
### Description of the bug

When adding connectors, only the male connector shows, there is no hole for the female?

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.1

### Operating system

Windows 10

### Printer model

Prusa MK4

lukasmatena, 2023-10-23:
Could you please try it 2.7.0-alpha1 and if you can reproduce it, post a 3MF? Thanks.

YuSanka, 2023-10-24:
Thanks for your report. However I believe that there is no issue with a Cut but with visualization of negative volume in selected object.
[image]

@rt6151 Did you try to slice your objects? I'm sure that everything works as expected.

rt6151, 2023-10-24:
Thanks!

I will try that out. However, if I might suggest, you might want to make the change in the software? Why show the pin (in the selected mode) and not show the hole?

Best,

Rick Thomas

YuSanka, 2023-11-08:
SPE-2022

lukasmatena, 2023-11-11:
Fixed in [2.7.0-beta1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.0-beta1). Closing.

closed 2023-11-11 by lukasmatena