prusa3d/PrusaSlicer #11499: revert "Improved user interface"

stevekenney318, 2023-10-22:
PrusaSlicer 2.7.0-alpha1:

Improved user interface
The visual looks of user interface controls has been improved. They are now more pleasant to look at in both light and dark mode and various UI quirks have been resolved (e.g. https://github.com/prusa3d/PrusaSlicer/issues/8877, https://github.com/prusa3d/PrusaSlicer/issues/6399).

[image]

**I personally do not see this as an improved user interface ...  I much prefer the compact spacing with the wider slider. Is there any way to switch it back without going back to a previous version ?**

DangerManiac, 2023-11-07:
Finally got around to using the new alpha and I agree, the drop-down menus are not an improvement, even the image above that is used to display the new drop-down menus looks better than what I see with actual use. At least let us be able to adjust the spacing like we can now with the font.

lukasmatena, 2023-11-11:
The combo boxes were further improved in [2.7.0-beta1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.0-beta1). Thanks for your feedback. Closing.

closed 2023-11-11 by lukasmatena

stevekenney318, 2023-11-12:
> The combo boxes were further improved in [2.7.0-beta1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.0-beta1). Thanks for your feedback. Closing.

Thank you very much ! It is much better now !