prusa3d/PrusaSlicer #11500: Bridges starting in the air since v2.60

FlintEastwood, 2023-10-22:
### Description of the bug

Bridges are starting in the air. They have no point to lay on.
Last working version is v2.52
[image]
[image]

### Project file & How to reproduce

[BridgeTest.zip](https://github.com/prusa3d/PrusaSlicer/files/13062928/BridgeTest.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

v.2.6.1

### Operating system

Win10

### Printer model

TwoTrees Sapphire-S

PaparazziN, 2023-10-22:
Hi @FlintEastwood!
Have you tried printing the attached and see how it comes out? During the development of PrusaSlicer 2.6.0 there were some changes made to how bridges are being handled (check the release notes for more info). So it might be that the behavior you are describing is expected. 

If the print fails, the issue you describe seem to be a duplicate of #10231 and there seem to be some discussion of whether that problem is fixed or not. I would suggest you add your findings there and close this issue and ask if the devs could reopen the original issue. If the model you attached is impossible to print that is.

FlintEastwood, 2023-10-22:
Hi @PaparazziN .
No it is not possible to print it that way. The bridges are falling down.
I don't understand, why this "top solid infill" is generated between the bridging parts.
And it starts with one "top solid infill" segment and then one bridging segment is started with nothing to glue on at the other side.

Using "Extra perimeters on overhangs" doesn't work, slicing comes out the same with one perimeter.
Instead it makes it worse when using 2 perimeters, as the bridge is made like a spiral in the air.

After reading #10231 it seems to be the same problem. Thank you for pointing to that.

Vadforean, 2023-10-25:
Great that this is reopened! Hope someone can locate what's causing it.

DoubleStrike, 2023-12-30:
I saw the related issue #10231 closed and this reopened. I don't see any updates on this for months. Any feedback from the developers at all?  This is an especially bad problem when printing with flexible materials, as it makes several kinds of prints pretty much impossible.

DovydasNavickas, 2024-03-05:
Would be nice to see some progress report if anything is being done with this or maybe more information is needed. This is a serious regression and I'm happy that we encountered this with a small print and didn't waste a lot of filament, but still took a chunk of our time to debug it. Having a problem on a bigger print would be detrimental and it makes the newer versions of PrusaSlicer simply dangerous to use while slicing.

lukasmatena, 2024-03-21:
Fixed in [PrusaSlicer 2.7.3-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.3-alpha1). Closing.

closed 2024-03-21 by lukasmatena