prusa3d/PrusaSlicer #11501: Feature Request: Update STEP import to import non-tesselated surfaces

WarrenSchultz, 2023-10-22:
Looking through the generated GCODE in 2.7.0-a1, I noticed that the fitted arc math misses a lot of curves, allowing for further optimization. I reimported the model as a STEP, and saw little to no difference. 

With the inclusion of G2/G3 in this alpha, it seems like the ideal time to refactor the import so that curved solid geometry is translated directly into GCODE arcs for highly optimized print files.

Thanks!

gudvinr, 2023-10-23:
Slic3r engine doesn't work with solid geometry and STEP files currently translated to mesh first. That's why you don't see any difference.

Check out #4989 for more info. This exact topic was discussed there.

WarrenSchultz, 2023-10-23:
Yes, I'm aware it is currently tessellated before use, I apologize if that wasn't clear. 
There are currently no settings for curve-fitting for G2/G3, and in the import process for STEP, there are no tessellation options. 
At the time that thread was originally posted, and updated by Lukas, arc support was not implemented on the printer firmware, which it now is. 
This current update of the firmware and PS to support smaller/more efficient gcode. Lukas' comment was geared around the quality aspect of tris vs. curves from an appearance standpoint, as I read it. If the goal of binary gcode and G2/G3 is optimization and smaller file sizes, being able to generate more arcs accurately is not necessarily low-hanging fruit, but something that gets a lot of benefit from better analysis of curves in the original model. Allowing control over curve-fitting would be a good first step, but adds a layer of complexity, whereas understanding native curves from the source model is more efficient and accurate.

lukasmatena, 2023-10-23:
> time to refactor the import so that curved solid geometry is translated directly into GCODE arcs

This "refactor" basically means to write the slicing core from scratch. The logic "STEP=arcs && GCODE=arcs` is by far not enough to make it work, given that the whole slicing process heavily depends on polygon operations. This will not happen anytime soon, if ever.

gudvinr, 2023-10-23:
FWIW, [there are](https://github.com/karl-nilsson/sse) slicing engines that use OCCT directly but they were written with solid geometry in mind.  
So there are sources where one could take an inspiration but it's very far from "pick and place".

WarrenSchultz, 2023-10-23:
Fair enough, thanks for your reply @lukasmatena 
Now that I'm more awake, I can also probably explain better my thoughts. 
I was more envisioning a change to the import process than refactoring the whole slicing engine.
i.e. on import of a STEP file, evaluate the shape for curved surfaces, and use that to change the tessellation to shapes that are optimal for detection as arcs. If that's overly complex, add a pop-up on import of STEP files that allows the user to customize tessellation properties.
Ideally, a visualization mode with a slider for adjusting curve-fitting for arcs/debugging the algorithm would be a nice feature, but that might be too much effort vs. reward.

closed 2026-09-29 by prusaslicer-bot[bot]