prusa3d/PrusaSlicer #11502: PrusaSlicer crashed when examining generated Gcode

tnxgam3rx, 2023-10-22:
### Description of the bug

After enabling Arc Fitting and reviewing a 3D file, the slicer crashed with 2 error messages that I could not replicate.

### Project file & How to reproduce

I prepared the part by splitting the item into parts and setting the Arc Fitting to "Enabled: G2/3 I J". The error message occurred when scrubbing through the 14th layer (at 2.80 mm) at line 93062, however I couldn't replicate the issue myself.
[Tolerance Test File V3.zip](https://github.com/prusa3d/PrusaSlicer/files/13063995/Tolerance.Test.File.V3.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.0-alpha1

### Operating system

Windows 11

### Printer model

Prusa MINI+

lukasmatena, 2023-10-23:
Thanks for the report, but I am also unable to reproduce it so far.

Jan-Soustruznik, 2025-04-02:
The reported problem cannot be reproduced on our side. We have checked with the latest release, version 2.9.1.The ticket has been closed due to inactivity. If the issue persists, please open a new ticket with the problematic file attached and detailed steps to reproduce. Thank you, have a nice day

closed 2025-04-02 by Jan-Soustruznik