prusa3d/PrusaSlicer #11503: PrusaSlicer crashed/freeze when opening step

okovjogurtu, 2023-10-22:
### Description of the bug

When opening step, PrusaSlicer freeze/crash, same on PrusaSlicer-2.6.1 and PrusaSlicer-2.7.0-alpha1

### Project file & How to reproduce

[pouzdro.zip](https://github.com/prusa3d/PrusaSlicer/files/13064305/pouzdro.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.1-linux / 2.7.0_alpha1-linux

### Operating system

Ubuntu 23.10

### Printer model

Voron 2.4 with Stealthburner

lukasmatena, 2023-10-23:
Thanks, reproduced. It looks that it hangs in OCCT when meshing the loaded solid from the STEP (where `BRepMesh_IncrementalMesh` object is being created). We will add it to our list.

AutomotiveLily, 2024-08-27:
Also experiencing the same problem when opening a specific .step file.

**Project file & How to reproduce**
Download and open, or use "Open in PrusaSlicer" button on file named "Silica container v18.step", at https://www.printables.com/model/65922-silica-gel-spool-container/files

**Version of PrusaSlicer**
Version 2.8.0+win64

**Operating system**
Windows 10 Pro, Version 22H2, OS build 19045.4780

**Printer model**
Elegoo Neptune 4 Plus with OpenNept4une software

closed 2026-09-29 by prusaslicer-bot[bot]