prusa3d/PrusaSlicer #11504: PrusaSlicer 2.6.x intermittently crashes while idle

Jens5395, 2023-10-22:
### Description of the bug

I frequently have open windows of PrusaSlicer sitting in the background. PrusaSlicer itself seems to be working fine but when just sitting idle, it will randomly crash (application closes by itself). I am running Ubuntu 23.04 and this has happened with all 2.6 versions and even the beta's. 
While the crashing is annoying, the really objectionable part is that after the crashing the computer will go into a screen blanking mode that blanks at 15 - 20 seconds of inactivity on the computer. I end up restarting Ubuntu to fix the issue and restore normal operation.

### Project file & How to reproduce

While a project is generally open, the crash has only happened while there was no interaction with PrusaSlicer. I have not been able to determine a reason for the crash - it seems to be totally random. Sometimes it takes days before it decides to crash.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.6.1+linux-x64-GTK2

### Operating system

Ubuntu 23.04

### Printer model

I have three printers, all using Duet controllers and the issue seems printer independant

lukasmatena, 2023-10-23:
It is very difficult to tell. The only thing I can suggest is to make sure that you have updated the graphics card driver, which is where I would suspect the issue might be.

Jens5395, 2023-10-24:
Thanks. I am pretty diligent about updating the computer as official updates become available. I don't like messing with pre-release drivers and I assume that any updates on the manufacturers end would likely be rolled into the Ubuntu upgrades fairly quickly. I suppose that might be wishful thinking. Ubuntu reports 'Intel XeGraphics (TGL GT2)'. I don't know if this graphics system has any known issues.

closed 2026-09-29 by prusaslicer-bot[bot]