prusa3d/PrusaSlicer #11505: 2.7.0-a1 scroll bars don't work on long dropdowns

WarrenSchultz, 2023-10-22:
### Description of the bug

When trying to use the scrollbar on long dropdown menus with the new UI, it is impossible to click on the bar itself to rapidly scroll to the bottom of the list.

[image]

### Project file & How to reproduce

1. Start a new project
2. If you don't already have numerous filament presets, use the wizard to add more
3. Go to the filament settings tab, and click the dropdown menu to select a filament
4. Attempt to click the scroll bar on the right to scroll to the bottom of the list. (See attached screenshot)

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.7.0-alpha1+win64

### Operating system

Win11 x64

### Printer model

Mk3S+

mihai-stanimir, 2023-10-23:
Same issue. All scrollbars look like this now. I have a long list of presets and scrolling with the mouse wheel is not smooth and too slow. Used to be much faster on 2.6.1.

GuyH77, 2023-10-25:
As above, this is now painful to navigate.

BaconFase, 2023-10-28:
Not only can you not grab/use the scroll bar, but the dropdowns act as if I was using a finger on a tablet/smartphone to scroll...but with a mouse. This is an awful change that should be toggleable if not removed.

Additionally, the dropdown list UI does not update until scrolling has slowed or finished. I have a mouse with infinite scroll. When I flick to infinite scroll the mouseover text will rapidly update, but the text list in the dropdown remains static.

https://www.youtube.com/watch?v=Nxt9vJnH5ps

lukasmatena, 2023-11-11:
Fixed in [2.7.0-beta1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.0-beta1). Closing.

closed 2023-11-11 by lukasmatena