prusa3d/PrusaSlicer #11511: Printer settings shows all possible printers even non-configured once

Jens5395, 2023-10-23:
### Description of the bug

[image]

In 2.6.1 (and all other previous versions, only the configured printers presets would show.

### Project file & How to reproduce

No project file involved. Simply go to 'printer settings' and hit the 'down' arrow to choose.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer Version:   2.7.0-alpha1+linux-x64-GTK3 Build:     PrusaSlicer-2.7.0-alpha1+linux-x64-GTK3-202310191626

### Operating system

Ubuntu 23.04

### Printer model

not relevant, 3 printers, all with Duet controllers

kocikdav, 2023-10-23:
Hello @Jens5395 ,
The version 2.7.0-alpha1 (or any other alpha release) does not share configuration with regular releases as 2.6.1. What should happen is that on the first run of the new release, you are prompted about exporting configuration from 2.6.1. There are several ways how unify your configurations. If you delete configuration folder of the application (PrusaSlicer-alpha) reachable by _Help -> Show Configuration folder_, next run of 2.7.0-alpha will ask about the export again. Note that deleting config folder is destructive operation and you could lose any user profiles stored there. 
Another way would be to go to Configuration Wizard and select your printers again. Then you can export your user profile from 2.6.1 and load them in 2.7.0-alpha, if you have any. 
Please note that there is different configuration folder for alphas, betas and final releases. Closing this issue. please reopen if any problem found.
[image]

closed 2023-10-23 by kocikdav

Jens5395, 2023-11-24:
This same issue still persists with the latest 2.7.0 release. The configuration that is used is the same as the configuration from my 2.6.x setup which works fine. I did also try to update the configuration making sure that only three printers are selected but I still get approximately 250 printer preset configurations listed when I run 2.7.0

Jens5395, 2023-11-24:
OK. doing some more digging and I found the reason for my troubles ....
When I start 2.7 I get the following error message:
Failed loading the preset file: /home/jens/.config/PrusaSlicer/printer/Jubilee - Copy, new.ini
Reason: Failed loading configuration file "/home/jens/.config/PrusaSlicer/printer/Jubilee - Copy, new.ini": 
Invalid value provided for parameter thumbnails: 0x0,0x0
- Input value is out of range

When I hit 'ok'  (after all, it seems to only affect one printer configuration), PS opens up with 'default fff' printer selected. When I then try to select a different printer I see the roughly 250 system presets.

If I delete the particular .ini file that causes the error and then run PS 2.7, I get no error message and only see the presets that were actually configured. I do not know how to edit or correct the .ini file that is corrupt/defective.