prusa3d/PrusaSlicer #11515: Add Basic Arrangement Option for objects that are all the same size.

BrienKing, 2023-10-24:
**Is your feature request related to a problem? Please describe.**
Yes.  Arrange doesn't work well with my small parts.

These should be arranged in a block for optimal layout.
[image]

**Describe the solution you'd like**
Add a Basic Arrange for parts that are of all equal size.   Equal being that they are easily bound by a box and not an odd shape that requires you to Tetris them on the plate.  This would apply to a lot of parts if you are just printing large quantities of them.
This arrange method would be very fast to calculate and execute.

**Describe how it would work**
The easiest solution is to allow me to specify the number of Rows and Columns to arrange the parts in.  If the parts exceed that then give an error.   Maybe only specify one, so I want 4 columns of the parts and it figures out how many rows it needs.

You could probably calculate the number of Rows and Columns based on the size of the object vs the size of the bed (minus print margins) and determine the optimal (or near optimal numbers)

Columns = (Bed Width - Margin) / WidthOfObject
Rows = (BedHeight - Margin) / HeightOfObject

From there you can calculate a ratio and get an estimate of how many columns you want based on the number of objects that would fit.

**Describe alternatives you've considered**
Manually arranging is a pain with small objects.   It's probably the OCD in me, but I like the parts to line up and doing it by hand is a real pain :)   (You could Add Snapping feature...)

Tupson444, 2026-09-18:
still relevant

Jan-Soustruznik, 2026-10-21:
A similar feature request already exists. This request has been merge to keep topics organized.

closed 2026-10-21 by Jan-Soustruznik