prusa3d/PrusaSlicer #11516: M600 color change applies at beginning of layer rather than when toolhead is picked up in that layer

AlmostFeral, 2023-10-24:
### Description of the bug

PrusaSlicer version 2.7.0-alpha1+win64 (Persistent since 2.6.1)

Steps to reproduce the issue:
Sliced a 5 color file using color change at layer height applied to the 4th toolhead to introduce a 6th color higher up in the model. 
Expected result: Printer would pause during the layer after the 4th toolhead had been picked up and would allow the printer to do a color change on that toolhead
Actual result: Printer pauses at the beginning of the layer while holding toolhead 3. Entering the tuning menu only allows a color change to commence on toolhead 3. 

### Project file & How to reproduce

[6 color Benchy.zip](https://github.com/prusa3d/PrusaSlicer/files/13093505/6.color.Benchy.zip)
[image]

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.0-alpha1+win64

### Operating system

Windows 11 Home

### Printer model

Prusa XL 5T

ex-nerd, 2023-10-28:
I think what I'm seeing is the same bug. Here's another `.zip` with 3mf project file and `.bgcode` exported from it, in this case for my 2-tool setup.

[toolchange_color_bug.zip](https://github.com/prusa3d/PrusaSlicer/files/13194657/toolchange_color_bug.zip)

After converting to text gcode, you can see that the `M601` (there's only one … line 71043) happens while still using `T0` and is several hundred lines of code before the next interaction with `T1` (where it heats up and then does the swap).

I would expect the `M601` to actually trigger after the tool has been heated and picked up, but before it moves to the wipe tower (since there is still almost always a few mm of extruded filament oozed out of the nozzle after a manual change and it would be nice if that didn't get stuck to the outside of a part)

Additionally (and maybe this is just a "feature"), the "sliced" UI layer slider doesn't actually show the selected layer change besides just a slight widening of the tick mark on the ruler … unless you're on the exact layer that triggers the pause:

[image]
[image]

ex-nerd, 2023-11-16:
I wonder if this is also related to single-tool color changes where the print head goes to "peck" at the spot it was before the color change (often leaving a drop of ooze with nozzles 0.6 and larger) before moving on to the new layer.

AlmostFeral, 2023-11-16:
> I wonder if this is also related to single-tool color changes where the print head goes to "peck" at the spot it was before the color change (often leaving a drop of ooze with nozzles 0.6 and larger) before moving on to the new layer.

I think it has to be something different. I know what you're talking about and I definitely see that on my non-prusa IDEX machine. I think I was seeing it on the pre-2.7.0-beta1 slicer with the XL as well, but it's been solved for the XL since the beta with the nozzle going to wipe on the purge tower before docking. I was so happy when I saw they had they added this that I PRANCED around my house.

ex-nerd, 2023-11-16:
> with the nozzle going to wipe on the purge tower before docking.

There is no purge tower or docking in single-nozzle multi-color prints, the tool head just moves between the print and the front left corner where the manual filament change happens. This happens all the time with Hueforge prints (I only have a 2-tool XL, so it's easiest to just use a single tool and swap colors manually). It looks like the `M600` gets added right before the tool is told to move between layers, so after the `M600` the nozzle is briefly returned to the last position of the previous layer, and then moves to the next layer.

Jan-Soustruznik, 2024-01-04:
Hello, @AlmostFeral ,

thank you for report. We will try to make a modification in some of next release.

ticked created: SPE-2098

lukasmatena, 2024-02-09:
Fixed in [2.7.2-alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.2-alpha2). Closing.

closed 2024-02-09 by lukasmatena