prusa3d/PrusaSlicer #11517: PrusaSlicer doesn't emit valid gcode for color change when it is set up between layers that use the selected tool

NerdOfEpic, 2023-10-24:
### Description of the bug

Problem:  Color change setup when tool being changed is between layers that use it will emit the wrong gcode.  (Also true for bgcode.)
Specifically:  If the tool is in use in a layer when a color change is set up, the slicer will emit an M600 (color change) correctly and the printer will guide the user through color switching as expected.  If the tool was used below the current layer and will be used again above the current layer and a color change is set up between those uses, the slicer will emit an M601 followed by an M117 instead.  This causes the printer to pause, does not display a message as expected, and which tool is active is somewhat random due to which tools are active in the selected layer.
In General: With multiple tools, it's reasonable to want to change colors for a single tool in an area of the print that is between two heights that use that tool.  For example, tool 4 is used for the first 25 layers, and again starting at layer 50, it makes sense to request a color change at something like layer 37 so it's between the two places it is used.
Possible solution:
In the case where a color change is configured for a tool in a part of the print where that tool is not in use (a situation common enough that I ran into it immediately when I sliced a 6 color model to try it).
This needs a slightly better color change gcode layout:

1. switch to the tool that is going to be swapped
2. unload the filament
3. reload the new filament
4. switch back to whatever tool was needed next

Otherwise the odds of the correct tool being selected are not good, and slicer seems to realize this situation already since it emits a different gcode when it encounters it.  Unfortunately what it emits doesn't work.

### Project file & How to reproduce

Slice a model that uses at least 1 color more than the number of tools being used.  Likely requires at least 2 tools to be in use. 
 Make sure the model has a section where the tool being color changed is unused between two areas separated vertically by a bunch of layers.  When you set it up this way the resulting gcode will include something like this:
 ;COLOR_CHANGE,T1,#000000
M601
M117 Change filament for Extruder 1

Instead of the expected:
;COLOR_CHANGE,T1,#00FF00
M600

But NEITHER is correct unless the code handles switching to the tool being changed before the M600 is emitted and switches back after.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.1 / 2.7.0-alpha1

### Operating system

Windows 11 (win64)

### Printer model

XL 5 Tool

AlmostFeral, 2023-10-24:
Thank you, this is much more eloquently and completely presented than in the report I tried to write up, #11516

Eddyg87, 2023-12-16:
Just wanted to put forward I just found this exact issue. And leaves you a bit stuck midprint... 

I put in 2 filament swaps in layers in the middle of where they were needed. I found the Gcode has put in an M601 to pause and an "M117 Change filament for Extruder 0". 

Not only does the printer not actually display the text when the M117 command is done there is noway to actually select the tool you want to swap filament on midprint. 

As stated above we need a tool swap, an M600 then a toolswap back to whatever the next tool that is required.

Jan-Soustruznik, 2024-01-04:
Hello, @NerdOfEpic , 

thank you for your report. We will check the problem. I closed this issue as duplicate of  https://github.com/prusa3d/PrusaSlicer/issues/11516

closed 2024-01-04 by Jan-Soustruznik