prusa3d/PrusaSlicer #11518: PrusaSlicer won't get past "Preparing Settings"

Norecane, 2023-10-24:
### Description of the bug

Recently did a fresh re-install of Windows 10, PrusaSlicer worked on the PC before reinstall, and nothing changed as far as hardware is concerned.

Attempting to Install PrusaSlicer 2.6.1

It will fully install but won't load past the Preparing Settings state in the startup. 

I have restarted and reinstalled about a dozen times. I have attempted to run both 2.7 and 2.6.0 which all run into the same issue. 

I have unplugged my monitors to a single monitor. 
I have unplugged my Corsair keyboard (K70).
I have deleted the line "window_mainframe"
I have made sure all Prusa files have been deleted before starting a new install.
I have ensured what drivers i can think to access are all up to date. 

the program WILL run in Safe Mode though. 

### Project file & How to reproduce

No file.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.1

### Operating system

Windows 10

### Printer model

Sovol SV06+

sliverbaer, 2023-11-30:
I am having the same issue.  Just installed Windows 11 and PrusaSlicer was the first thing I installed.  It was working great.   Then I installed a few other things like Corsair iCue.  Now stuck on "preparing settings tabs..."

[image]

[image]

**Project File**:  None.  Just opening PrusaSlicer from shortcut.

**Version**: 2.7.0+win64

**OS**:  Windows 11 23H2

**Printer models**: MK3S+, KP3S, KP3S Pro S1

SolarSupremacy, 2023-12-07:
Would be nice if Prusa fixed this issue. Been happening for the last 4+ years.

Mine ran fine with my Corsair devices but now Prusa won't run at all. Just softlocks.

TobbeJ, 2024-01-14:
i also got this issue with stuck att "Preparing settings tabs..."
i suspect it is related to my corsair 12 button mouse.
i have had this problem for a very long time and most of the times prusa slicer gets stuck at the splash screen for a few minutes before it decides to start.

closed 2026-09-29 by prusaslicer-bot[bot]