prusa3d/PrusaSlicer #11519: SVG is just a squared cube

stritti, 2023-10-24:
### Description of the bug

I wanted to try the new feature to use SVGs as modifiers.
I added a logo in several different SVG layouts, but I could only see the emblem, but it will not be used for the layers. there is just a qube.

i tried several versions of SVG, but with no success. I add project file and the SVGs.

[image]
[image]

### Project file & How to reproduce

[svg-test.zip](https://github.com/prusa3d/PrusaSlicer/files/13112054/svg-test.zip)

I think you have to reimport SVGs because of path. I stored project in same directory to probably prevent this.
I tried to add a qube and several versions of SVG as own form or as modifier of the added generic qube.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.7.0-alpha1+win64

### Operating system

Windows 11

### Printer model

Prusa MINI+

Jony01, 2023-10-24:
https://github.com/prusa3d/PrusaSlicer/issues/11506

closed 2023-10-24 by Jony01

stritti, 2023-10-24:
> #11506

oh, thanks. Haven't seen this issue. Would be nice, if it would be possible to select or remove some areas from the imported SVG. Probable something like "ungroup SVG"?