prusa3d/PrusaSlicer #11528: [Enhancement] Move extruder to next color swap destination before M600

rcmaniac25, 2023-10-25:
**Is your feature request related to a problem? Please describe.**
Color swaps are a very popular way to make multi-color prints easily. One issue I've noticed plenty of times though, is the new color getting a tiny dot in a spot where it's not supposed to be. This is due to filament "drooling" from the hot nozzle after being being primed during the M600 command... but the extruder returning to the prior location.

Basically, extruder follows the following process:
1. Move to (X1,Y1), the last position before the new layer
2. Retract and raise to Z1+layer height+z-hop height
3. Move to (X2,Y2)
4. Lower to Z2+layer height and unretract

And when a color swap is added, it becomes:
1. Move to (X1,Y1), the last position before the new layer
2. Retract and raise to Z1+layer height+z-hop height
3. Do color swap operation (M600) in whatever manner the firmware is programmed to do.
4. Move to (X2,Y2)
5. Lower to Z2+layer height and unretract

What happens is that the filament gets primed and starts drooling, the extruder moves back to (X1, Y1, Z1+layer height+z-hop), and a tiny dot of filament is left on the print... it then moves the (X2, Y2, Z2+layer height) and continues printing like normal. If lucky, no dot will be left. If unlucky, not only will a dot be left, but a string of filament will be left too between the 2 points.

Before M600 (the last spot the extruder is before doing the color swap. Note: PrusaSlicer preview does extra operations like wipes and doesn't let you see the gcode preview of the exact end of the layer)
[image]

After M600 (where the extruder is after it does the move from the last location that prints occurred)
[image]

**Describe the solution you'd like**
I would like to request the extruder get moved to the destination location before M600. To use the list before:
1. Move to (X1,Y1), the last position before the new layer
2. Retract and raise to Z1+layer height+z-hop height
3. Move to (X2,Y2)
4. Do color swap operation (M600) in whatever manner the firmware is programmed to do.
5. Lower to Z2+layer height and unretract

**Describe how it would work**
M600 saves the current XYZ, moves to a pre-designated spot ([sometimes poorly](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3253)), unloads and reloads filament. Then (at least for Prusa printers) it does a priming action to ensure the filament fills the extruder. At this point, it returns to the saved XYZ spot and continues printing by traveling to the next print location. The priming and the return to the original XYZ location result in the dot.

**Describe alternatives you've considered**
Given the printer requires attention to do a filament swap, I usually have pliers and a small pick. The extruder returns to the original XYZ, then continues printing at the next print location, and if I see a dot of filament, I try to remove it with the pick.

neophyl, 2023-10-25:
Basically a duplicate issue.  Please see https://github.com/prusa3d/PrusaSlicer/issues/5195  or https://github.com/prusa3d/PrusaSlicer/issues/5951  , or https://github.com/prusa3d/PrusaSlicer/issues/4432 or https://github.com/prusa3d/PrusaSlicer/issues/2672 

I'm sure there are others too.  Please check for existing issues before opening new ones.

rcmaniac25, 2023-10-25:
Already did. #5195 and #5951 are not related as they are literally the requests for swapping filament, not changing how the swap works. #4432 and #2672 did not show up in my search before hand.

Don't make assumptions that a search wasn't done ahead of time.

neophyl, 2023-10-25:
The first 2 are related, the end result would be allowing the user to change the swap point of the filament change and avoid the visible dot of colour.  Even the dev's who monitor this github think so given the issues linked/mentioned.  

As to search, my apologies for assuming you hadn't searched.  It was a reasonable assumption given I found multiple ones with 1 search key in under 5 minutes. 
When you see just HOW many issues raised here are duplicates you will get to the point of making that same assumption as it is true in the majority of cases.

lukasmatena, 2024-02-18:
Solved in 2.7.2-alpha2. Closing.

closed 2024-02-18 by lukasmatena