prusa3d/PrusaSlicer #11529: New UI list scrolling is very slow

rcmaniac25, 2023-10-25:
### Description of the bug

In PrusaSlicer 2.6.1, with all filaments enabled, I can scroll the dropdown of filament presets in 2 scroll wheel uses.
In PrusaSlicer 2.7.0 alpha1, that same setup and operation takes 46 scroll wheel uses.

### Project file & How to reproduce

Open PrusaSlicer and any of the preset dropdowns that require you to scroll will have the issue.

While the request is to fix the scroll speed, as there is a new UI elements, it would be useful to allow dragging the scroll bar too. But that may be a request as opposed to a bug in the new UI.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.7.0-alpha1+win64

### Operating system

Windows 10 Pro 22H2

### Printer model

Not applicable

lukasmatena, 2023-11-11:
Fixed in [2.7.0-beta1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.0-beta1). Closing.

closed 2023-11-11 by lukasmatena

rcmaniac25, 2023-11-14:
@lukasmatena it is better, though now at 6-8 scroll wheel uses. It appears that there may be a coasting animation or such being used in the background (to produce the effect of smooth scrolling when scrolling quick) that doesn't actually work. So it jumps around and is jittery. But at least it's usable again... just could use more work to feel like a functional UI element.

Shmidtey, 2023-12-10:
This is still an issue for me on the stable release of 2.7.0. on Windows 10 pro.
(slow/laggy/less sensitive scrolling of dropdown menus/lists for print, filament, and printer settings)

However I'm afraid my problem may be an edge case due to being caused by interactions with Prusaslicer and my particular monitor/windows display settings. Sharing here anyway in case my findings help someone.
Monitor is an Asus ROG swift PG43UQ 3840x2160, windows scaling and layout 150% (scaling due to high dpi display causing teeny tiny icons and UI text)

It only happens with windows scaling and layout set between 100% and 200%, and with Prusaslicer in advanced or expert mode. 
Switching to simple mode and closing/reopening the app resolves the problem, but only while the app remains in simple mode. Further closing and re-opening the app in simple mode retains the desired scrolling behavior (fast/snappy/sensitive).

Temporary workaround for faster scrolling in advanced/expert mode: changing application text size in the preferences restarts application, and then scrolling is fast again. However this is temporary and the app only retains the desired scrolling behavior until you close the app and reverts once re-opened. 
If you open additional app instances the workaround will have to be applied to each instance.

Unfortunate since I always prefer to leave my app in expert mode, and that my icons are too small at 100% scaling and too large  above 200% scaling. 
200% does work and I'll probably set it there , but its a little large for my liking.

rcmaniac25, 2023-12-23:
Hmm... I am also running 4K with 150% scaling (3x of them with extended desktop). I tried simple mode and adjusted text size and did not get an improvement in scrolling. I did not try 100% or 200% scale (I have a lot of applications open at the moment and I know changing scale causes them to repaint their UI and some don't like such a drastic change and crash, so for the sake of my time, I'll wait to try a different scale for when Windows decides it has updates and needs me to restart)