prusa3d/PrusaSlicer #11533: Stuttering in XLIS

mdjurfeldt, 2023-10-25:
### Description of the bug

The extruder "stutters" in sections of a perfect circle with visible artefacts.

https://github.com/prusa3d/PrusaSlicer/assets/3612425/32d16231-c058-4858-a889-9856cde0b173

### Project file & How to reproduce

[dryer-XL-Body.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/13168748/dryer-XL-Body.3mf.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.7.0-alpha1+linux-x64-GTK3

### Operating system

Debian Bookworm

### Printer model

5-tool XL

rtyr, 2023-10-26:
>  sections of a perfect circle 

Looking at the model, it is not a circle at bottom area.

[image]

mdjurfeldt, 2023-10-26:
Ah... sorry. I should always look at the model first. That bug in the model was unexpected. Closing this issue.

closed 2023-10-26 by mdjurfeldt