prusa3d/PrusaSlicer #11534: Introduce new printer profile parameter "On Soluble Infill"

mdjurfeldt, 2023-10-26:
**Is your feature request related to a problem? Please describe.**
In Issue #214 (which is acted on and closed), I complained about the speed 60mm/s being used for Solid infill in soluble profiles since it induces resonances in the XL. I thought it could be set to 80mm/s (or higher) but it was explained to me that this parameter is *also* used when infill is printed on top of soluble support and that, for this reason, it can't be too high.

This seems a bit crazy to me. Solid infill is being printed quite a lot in a model. For most of the infill you want this high, but when printing on soluble material you probably want it pretty low (actually lower than the current 70mm/s.

**Describe the solution you'd like**
I would like you to introduce a new parameter under Printer Settings called, for example, "On Soluble Infill" which is used when solid infill is printed on top of soluble materials. Separating this parameter from Solid Infill makes it possible to optimize speed for printing infill in a model and speed for printing on top of soluble material separately.

mdjurfeldt, 2023-10-26:
Moved this to PrusaSlicer-settings.

mdjurfeldt, 2023-10-26:
Reopening again: I realized that this is actually a change in PrucaSlicer rather than settings. Leaving the old issue https://github.com/prusa3d/PrusaSlicer-settings/issues/215 open.

rtyr, 2023-10-26:
I will close this. I moved it from PrusaSlicer-settings.

closed 2023-10-26 by rtyr