prusa3d/PrusaSlicer #11535: [Feature request] Introduce new Printer Settings parameter "On soluble infill"

mdjurfeldt, 2023-10-26:
In Issue https://github.com/prusa3d/PrusaSlicer/issues/214 (which is acted on and closed), I complained about the speed 60mm/s being used for Solid infill in soluble profiles since it induces resonances in the XL. I thought it could be set to 80mm/s (or higher) but it was explained to me that this parameter is also used when infill is printed on top of soluble support and that, for this reason, it can't be too high.

This seems a bit crazy to me. Solid infill is being printed quite a lot in a model. For most of the infill you want this high, but when printing on soluble material you probably want it pretty low (actually lower than the current 70mm/s.

I would like you to introduce a new parameter under Printer Settings called, for example, "On Soluble Infill" which is used when solid infill is printed on top of soluble materials. Separating this parameter from Solid Infill makes it possible to optimize speed for printing infill in a model and speed for printing on top of soluble material separately.

tg73, 2024-02-07:
This also applies when using dissimilar materials for the part and the support (eg, PETG and PLA). Solid infill on the first part layer on top of zero-z-gap dissimilar support material currently goes too fast and can fail.

Also, the naming of "SOLUBLE" - both in terms of print settings and as the moniker for zero-z-contact-distance supports is not adequately inclusive of non-soluble dissimilar materials. Non-soluble dissimilar materials appear to be increasingly popular with XL multi-tool users, many of whom (myself included) have never used and are never likely to use soluble support material. Due to the naming, I've never considered selecting a "SOLUBLE" profile because I'm not using soluble material. Also, the SOLUBLE profiles don't include eg 0.6mm nozzle 0.32 layers. I might well want zero contact distance dissimilar material supports with the 0.32 SPEED profile.

mdjurfeldt, 2024-02-28:
What about "On interface"?

A note of caution here: This does not just apply to "Solid infill". I just had a situation where the feature printed as the upper interface (i.e. printed *on* the interface of a different material) was not Solid infill but Perimeters. This makes this issue a bit complex: Should, e.g., Perimeters printed on an interface be printed at a different speed than Solid infill printed on an interface?

Maybe one solution would be to still just have one extra parameter "On interface". If this is an absolute number, it will be used regardless of feature. If it is a percentage, the speed will be computed as this percentage of the speed of the feature being printed.

mdjurfeldt, 2026-09-18:
I still want this feature. It would definitely be a major improvement to be able to set speeds separately for the first layer on top of a foreign material. (Caveat: Not 100% sure if recent improvements contain corresponding functionality.)