prusa3d/PrusaSlicer #11539: Dynamic/variable line speed

qaqa339, 2023-10-27:
Is any chance to make dynamic line speed like _Dynamic overhang speed_ but for straight lines by some treshold?
Let's say, i have large cubic object with vertical holes. For straight 'outer' perimeters is high printing speed possible and wanted, but holes are curved, there i want much slower speed for better result, but they are printed same as perimeter speed. So i must set all perimeter speed safe for curved/short lines. And that is slooow, so slooow...

Example: line length above 10mm = 100% speed, below 10mm = 60%, below 5mm = 30%

rtyr, 2023-10-27:
Such feature does not make much sense to me. You can set low print acceleration to achieve similar effect.

Separate speed setting for selected areas (without breaking perimeters) may be usefull in same cases, but we already have issues on that topic, for example https://github.com/prusa3d/PrusaSlicer/issues/3370.

qaqa339, 2023-10-28:
I agree, this is not feature for printing toy figures. My goal is printing straight long lines as quick as possible, and simultaeously short lines as good as possible. Acceleration can handle this to some degree, but it is not 'right nutty' ;)
Thanks anyway.

closed 2023-10-28 by rtyr