prusa3d/PrusaSlicer #11540: 2.7.0-alpha1 crashes when saving large project with SVGs and Texts

FloSchmidt, 2023-10-27:
### Description of the bug

Prusaslicer crashes (maybe not always, I will dig deeper) saving a project with multiple (16, but previous crashes where with less) Objects.

Only have the G-Code left so I attatched a screenshot of the G-Code to show what I actually wanted to save... 
[image]

Here is the last dump from wich i did the screenshot.
[prusa-slicer.exe(1).30716.dmp](https://github.com/prusa3d/PrusaSlicer/files/13187980/prusa-slicer.exe.1.30716.dmp)

These three dumps whrere caused by the same issue, but there where not as many objects on the buildplate. Sorry I don't remember exactly. I hope they help anyways.
[prusa-slicer.exe.5804.dmp](https://github.com/prusa3d/PrusaSlicer/files/13187970/prusa-slicer.exe.5804.dmp)
[prusa-slicer.exe.12496.dmp](https://github.com/prusa3d/PrusaSlicer/files/13187990/prusa-slicer.exe.12496.dmp)
[prusa-slicer.exe.31456.dmp](https://github.com/prusa3d/PrusaSlicer/files/13187493/prusa-slicer.exe.31456.dmp)

BTW: I absolutely love PrusaSlicer and the MINI! Great work!

### Project file & How to reproduce

A I cannot save the project I only can attatch the source files.

The object is from https://www.printables.com/de/model/510422-bottletag-v2/comments/1226231
SVG is [image].
Text was in Consolas, 10mm 0x{twoHexDigits}

[image]

Copy the master around and try to save the project.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.0-alpha1

### Operating system

Windows 10 22H2 (Build 19045.3570)

### Printer model

Prusa MINI

Jan-Soustruznik, 2023-11-03:
SPE-1998

Jony01, 2023-11-13:
Should be fixed with [2.7.0-beta1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.0-beta1)

closed 2023-11-13 by Jony01

FloSchmidt, 2023-11-13:
Great News, thank you all!