prusa3d/PrusaSlicer #11541: No G-code preview/weird text when sending gcode file from latest PrusaSlicer alpha to latest MK4 firmware alpha

RM-Commando, 2023-10-27:
### Description of the bug

After downloading latest alpha of PrusaSlicer and latest Alpha of Mk4 firmware, when I sent a g-code (not binary) file straight from prusaslicer to my mk4 (button on bottom right of prusaslicer), there was no G-code preview and some glitchy/abnormal text on the screen
[image]

### Project file & How to reproduce

1. Duplicate settings (in 3mf and screenshotted in .zip)
2. slice (prusaslicer 2.7.0 alpha1 win64)
3. send to mk4 printer (on firmware 5.1.0 alpha2) using send g-code button (just send, dont start print)
4. view on mk4 display 

[weird gcode preview.zip](https://github.com/prusa3d/PrusaSlicer/files/13189023/weird.gcode.preview.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

 2.7.0 alpha1 win64

### Operating system

Windows 11

### Printer model

Mk4 on firmware 5.1.0 alpha2

rtyr, 2023-10-27:
New firmware supports only preview in QOI format, while your custom printer profile (looks like a copy of some older system profile) is configured to generate PNG format, which is no longer supported by new 5.1.0 firmware.

From 5.1.0-alpha2 release notes https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.1.0-alpha2.

> 
> QOI instead of PNG (XL, MK4, MINI)
> 
> Until this release, the thumbnails for G-code were in PNG format, which was rather resource-heavy. We have decided to switch to another format called QOI, which is way less demanding, and we can utilize the freed-up resources to bring more features in future releases.
> 
> The only downside of removing the support for PNG is that all the G-codes sliced until now, won't have a visible thumbnail on firmware 5.1.0 or newer. For now, the only solution is to reslice the G-code in PrusaSlicer 2.7.0 or newer, however, we are investigating other options.

closed 2023-10-27 by rtyr

RM-Commando, 2023-10-27:
How would I change it to use QOI? The gcode was sliced in 2.7.0

rtyr, 2023-10-27:
> The gcode was sliced in 2.7.0

Yes, but with not up-to-date printer profile.

You can either use standard printer profile included in 2.7.0 or modify your custom profile like this:

[image]