prusa3d/PrusaSlicer #11544: When uploading files to prusa connect whilst Mk4 is printing, printer pauses stating, "USB drive or file error... Reconnect the drive"

RM-Commando, 2023-10-27:
### Description of the bug

I was printing something on my Mk4 and wanted to add a file to the queue in Prusa Connect. uploading the file to prusa connect went fine but going from prusa connect to mk4 whilst printing didn't work (the upload to printer started automaticly). It didn't work beacuse the printer would pause the current print multiple times during the upload with the pictured message. I'm not sure if this is a bug or a safety feature but I thought it was worth reporting anyway.
[image]

### Project file & How to reproduce

1. slice file in PrusaSlicer 2.7.0-alpha1+win64 (not sure if this is relevant)
2. upload sliced **_Binary G-code_** to prusa connect.
3. after file is uploaded to prusa connect it will automatically begin sending/uploading to printer (in my case Mk4 running 5.1.0-alpha2)
4. see on screen that the print has has and pictured message is displayed
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer 2.7.0-alpha1+win64

### Operating system

Windows 11 x64

### Printer model

Mk4 running 5.1.0-alpha2

zuzu1383, 2023-10-28:
I am having identical problem. Will avoid transferring files while printing till this gets fixed

Jan-Soustruznik, 2023-11-01:
Hello, @RM-Commando,

thank you for your report. We investigated your reported problem with Firmware team and found that the problem isnt caused by Slicer, but printer firmware. Fix of this issue should be implemented in new FW release. 

I closed this issue as solved, but If it still remain with new FW release, then create issue dirrectlly to FW team please, at link here -> https://github.com/prusa3d/Prusa-Firmware-Buddy/issues

Thank you, have a nice day

closed 2023-11-01 by Jan-Soustruznik