prusa3d/PrusaSlicer #11547: .3mf files with similar sized components aren't being defined as multi-part objects

CoryatJohn, 2023-10-28:
### Description of the bug

I have an assembly in Fusion 360 that consists of two identically sized parts stacked on top of each other. I exported the model as a .3mf and when importing the file into prusaslicer, it fails to ask the question regarding multi-part objects. The result is Prusaslicer loads the parts flat on the plater instead of stacking them.

If I change the size of one of the parts, Prusaslicer correctly identifies the assembly and asks the multi-part question.

I'm including two projects: one that imports correctly and one that has the error. The difference between the two projects is the height of the parts. In the error, both parts are the same height but in the project that imports correctly, the parts are different heights.
[Multi-material test.zip](https://github.com/prusa3d/PrusaSlicer/files/13194233/Multi-material.test.zip)

The point of this project is to test compatibility of various filaments to each other using my XL with two tools.

Thanks!

### Project file & How to reproduce

To reproduce the error, import the "Project with error" 3mf file and see that two distinct objects are created. Importing the "Project imports correctly" file shows the correct stacking of the objects.

The only difference between the two files is that the project with error file has two identical objects stacked on top of each other. One of the objects has the text "Test 2" cut into it. 

The project that imports correctly has two parts with different heights but otherwise identical. 

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

 PrusaSlicer-2.6.1+win64-202309060711

### Operating system

Windows 10 (build 19045), 64-bit edition

### Printer model

Prusa XL multi-tool

gegy1, 2023-12-13:
I have a similar issue: Modelt a part in Solidworks, saved as 3mf and imported it for printing in a prusa slicer XL (2 head) profiled project. I am getting the dialog asking for multi material (multi body) part. Then i made a copy of the part in Solidworks, modified some dimension and saved it as another 3mf file. Now prusa slicer does not show the dialog and imports the two bodys as two objects. 
Also the two objects are placed not together on the build plate. 

Workaround: Save the file as STL, import in Pruca slicer and split into parts (using the "split to parts" tool from top toolbar)

I have added the two 3mf files and the STL file

Left-Front.3mf -> shows the dialog
Einlage-Teil-1.3mf -> Does not showing the dialog for multi bodies
Einlage-Teil-1.STL -> Can be imported and split into parts (expected result for the 3mf)
[Attachments.zip](https://github.com/prusa3d/PrusaSlicer/files/13664004/Attachments.zip)

YuSanka, 2024-01-11:
Thanks for your report.
SPE-2104

mperino, 2024-01-17:
Thanks @neophyl yea sounds like my issue is the same. And we both seem to have run into it the same way. Making tests for 2 tool XL. 
Can probably be merged into this issue.

FidelCapo, 2024-02-01:
This issue should be fixed and the fix will be part of the next release.
Closing.

closed 2024-02-01 by FidelCapo