prusa3d/PrusaSlicer #11550: Custom Variables for Custom GCode

yonitjio, 2023-10-28:
**Is your feature request related to a problem? Please describe.**
I'm using my own approach to shape the filament tips, so I use custom tool change gcode. But I need values that are specific to each filaments. Currently I use variables that are used only for statistics, e.g. density, cost and weight as a hack.

**Describe the solution you'd like**
Some kind of array of name-value pairs on filament settings tab, so I can refer as filament_custom_variable["my variable"][next_extruder] instead of filament_cost[current_extruder].

Or a simple array with index also would suffice, filament_custom_variable[0][next_extruder].

**Describe how it would work**
I have no idea.

**Describe alternatives you've considered**
None

**Additional context**
None.

gudvinr, 2023-11-02:
2.6.0 has new macro language capabilities that look like what you want:
https://github.com/prusa3d/PrusaSlicer/wiki/PrusaSlicer-Macro-Language

yonitjio, 2023-11-03:
What I'm suggesting is for the UI, some kind of table UI. The values then can be used in the macro.

closed 2025-09-06 by yonitjio