prusa3d/PrusaSlicer #11551: Improve vertical print accuracy on ovehangs

vido89, 2023-10-28:
So I set thickness of overhang to be 0,5mm on left side of picture and 0,4mm right side but slicer splits it in to 5 layers so 5 layers time 0,2 mm thickness gives me over 1mm after printed 

**Describe how it would work**
The slicer should be giving me 2 or 3 vertical layers  

**Additional context**
[image]

xepls, 2023-10-28:
You have a problem (I belive), you used 0.2 mm layer height and you want a resolution of 0.1 mm. The program in this case choose for 0.3mm or 0.5 mm to this overgang in 0.4 mm.

You can fix if you choose a layer height more detail (0.1 mm / 0.16 mm or lower)

xepls, 2023-10-31:
> You have a problem (I belive), you used 0.2 mm layer height and you want a resolution of 0.1 mm. The program in this case choose for 0.3mm or 0.5 mm to this overgang in 0.4 mm.
> 
> You can fix if you choose a layer height more detail (0.1 mm / 0.16 mm or lower)

I understand bad.

Check if top solid layer is check. is posible that you had set in 1 mm

neophyl, 2023-11-01:
As well as what has been pointed out above about layer height selection from your preview picture it looks like you have thick bridges enabled.  When that legacy setting is on the layer height of bridges is not based on your selected layer height but on nozzle width.  So for a 'normal' 0.4mm nozzle your layer height will be double your set 0.2.  This will be exacerbating the issue.

You would have to attach your zipped up project file to the issue as requested when opening one to confirm this though.

closed 2026-09-29 by prusaslicer-bot[bot]