prusa3d/PrusaSlicer #11552: Support filament settings per tool

ccroswhite, 2023-10-28:
**Is your feature request related to a problem? Please describe.**
Currently, with an XL multitool, there can only be a single setting for filament.  This causes problems when either printing with different type filaments or when using different manufacturer filaments that require different settings like nozzle temperature or retraction speed.

**Describe the solution you'd like**
Aside from the UI/UX of adding support for each tool (in my case 5), I would like the ability for each tool to have independent settings

**Describe how it would work**
Selecting a "tool" (1-5), would render an additional a separate filament settings "page".  All the settings then would be associated with the tool and the associated filament in that tool.

**Describe alternatives you've considered**
Aside from the UI/UX design, not sure there are any alternatives to this feature.

**Additional context**

rtyr, 2023-10-28:
I am not sure what do you mean, you can already select different filament profiles (with different settings) for each tool.

rtyr, 2023-10-28:
[image]
[image]

rtyr, 2023-11-23:
No response, possible duplicate of https://github.com/prusa3d/PrusaSlicer/issues/11613. Closing.

closed 2023-11-23 by rtyr