prusa3d/PrusaSlicer #11553: XL: Sickering constantly before and while MBL

gbzh01, 2023-10-29:
### Description of the bug

When the XL is preparing the MBL, there is constantly filament running out of the nozzle. Can you find a solution against it?[](url)
[image]

### Project file & How to reproduce

Start a print, here it's PETG

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.0 alpha1

### Operating system

Mac OS 14.0

### Printer model

Prusa XL

GuyH77, 2023-11-01:
They just need to drop the temp, 170 is too hot on the XL. It leaves small bits of filament all over the bed. Agree with the OP, this needs to be fixed.

rtyr, 2023-11-17:
The temperature was now lowered from 195C closer to the previous value (currently 175C), it should help with the oozing you are refering to (unless the filament have a lot of moisture in it). The temperature needs to be high enough to not cause issues with load cell.

closed 2023-11-17 by rtyr