prusa3d/PrusaSlicer #11555: Prusaslicer 2.6.0 organic supports html links invalid

Will4567, 2023-10-29:
### Description of the bug

In prusaslicer 2.6.0, in the Organic Supports section of the settings (which is in support material section which is in print settings), all the HTML links that are enabled for hovering and clicking on the settings, lead to things like:

https://help.prusa3d.com/en/article/support-material_1698#tree_angle

This is incorrect, the organic supports have moved to a separate organic supports webpage:
https://help.prusa3d.com/article/organic-supports_480131
All those links need to be updated.

Also, everyone one of the # shortcuts has "tree" in the name, shouldn't they be called organic, not tree? :)

### Project file & How to reproduce

[prusabug1.zip](https://github.com/prusa3d/PrusaSlicer/files/13198620/prusabug1.zip)

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0

### Operating system

windows 10

### Printer model

prusa mk3s+

Jan-Soustruznik, 2023-11-03:
SPE-2014

lukasmatena, 2023-11-11:
Fixed in [2.7.0-beta1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.0-beta1). Closing.

closed 2023-11-11 by lukasmatena