prusa3d/PrusaSlicer #11556: New orange highlight makes selected menu item hard to read

dwineman, 2023-10-30:
### Description of the bug

The UI controls have been revamped in 2.7.0 to have different spacing and an orange highlight when selected:

[image]

However, the orange highlight makes the selected item much more difficult to read than before, especially in dark mode on MacOS, where the text is white:

[image]

This color combination fails the [WCAG contrast checker](https://webaim.org/resources/contrastchecker/) at this text size:

[image]

In the previous release, the system accent color was used, in accordance with the MacOS human interface guidelines. The reason for this guideline is that the accent colors have been chosen for contrast and readability against dark and light text colors, avoiding a whole class of accessibility problems when used correctly. For example, here's PrusaSlicer 2.6.1 with the orange accent color selected in System Settings, and it's perfectly readable, because the OS adjusts UI colors for readability:

[image]

Forcing the UI highlight color to a brand color is an accessibility regression, not to mention an amateurish mistake, and it harms your users who do not have perfect vision (which is ALL OF THEM, eventually). Please revert this change and don't ever do this again.

### Project file & How to reproduce

UI/accesibility issue that affects all projects and configurations. I'm checking the box for Project File so that this gets looked at, but there is no need for a file.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.0-alpha1+ MacOS-×64

### Operating system

MacOS 13.5.2

### Printer model

MK4

lukasmatena, 2023-11-11:
The highlight was changed to grey in [2.7.0-beta1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.0-beta1). Closing.

closed 2023-11-11 by lukasmatena

dwineman, 2023-11-12:
Thanks, this is an improvement, but there are now two new problems: 1) in Dark Mode on MacOS, the highlight is now very hard to see because it's too similar to the background gray:

[image]

2\) When Dark Mode is toggled on or off while PrusaSlicer is running, the highlight color doesn't update, resulting in very poor legibility. I'm certain that this is because [Apple's system UI colors handle the change automatically](https://developer.apple.com/design/human-interface-guidelines/color#System-colors), but they're no longer being used as of 2.7.0, so additional code is required.

[image]
[image]

Please just revert to the previous highlighting code, which used the system accent color correctly and avoided all of these problems that have already been solved by the platform vendor.