prusa3d/PrusaSlicer #11557: Parts list slowing down interface

krisrok, 2023-10-30:
### Description of the bug

If there are more than just a few parts loaded, the parts list on the right side gets very unresponsive. This also affects object picking in the 3D view.

On a 4K screen with high dpi scaling active, object picking or scrolling through the parts list is very unresponsive, see this recording:
<video src='https://github.com/prusa3d/PrusaSlicer/assets/3404365/c76db621-ff40-40d1-a5b6-02d66322481a'/>

Note that on the end of the video I'm switching from "Expert" to "Simple" view and object picking works very well there even though the number of parts is still the same.

If I resize the window's height to render a smaller parts list, it gets more responsive quickly. But it still is sluggish.

### Project file & How to reproduce

I've opened the [3mf](https://files.printables.com/media/prints/618300/stls/4905261_c75cceec-6afd-4384-8f07-09b04268fae0_d1db1a29-2654-4457-a1f7-3561534c9db6/spider-all-parts.3mf) from the [spider model](https://www.printables.com/model/618300-articulated-spider).

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0 to 2.7.0-alpha1

### Operating system

Windows 10

### Printer model

MK3

kubispe1, 2023-10-31:
Hi, thanks for report. There is a note from @Stefankaa I want to mention. 
https://github.com/prusa3d/PrusaSlicer/issues/10996#issuecomment-1724158575

> The 2.5.0 release works great

krisrok, 2023-10-31:
> Hi, thanks for report. There is a note from @Stefankaa I want to mention. [#10996 (comment)](https://github.com/prusa3d/PrusaSlicer/issues/10996#issuecomment-1724158575)
> 
> > The 2.5.0 release works great

Indeed, the problem isn't present with 2.5.0. Picking and scrolling is all snappy:
<video src='https://github.com/prusa3d/PrusaSlicer/assets/3404365/ffe03c12-dc74-4e90-893e-d9d20f0a947c'/>

kubispe1, 2023-11-01:
SPE-2001

cupradito, 2023-11-02:
Hello.

Same here.

Fine with Prusa slicer 2.5 , 4K High dpi display.
Bad with prusa slicer 2.6 and 2.7

krisrok, 2023-11-28:
There's a (kind of shitty) workaround I found with 2.7.0 and wanted to share: You can use the "collapse/expand" button to temporarily hide the sidebar e.g. if you just want to be able to quickly arrange a parts by hand:

[image]

There's no UI lag with the sidebar hidden.

You could also switch to the "Simple" sidebar tab but it takes 3-4 seconds (!) to switch from "Expert" or "Advanced" to "Simple" in my test case. Using the "collapse/expand" button is way quicker.

(It's almost certainly a performance problem inside the parts list item rendering. Collapsing some of the list's foldouts reduces the number of rendered list items and drastically improves performance as well.)

closed 2026-09-29 by prusaslicer-bot[bot]