prusa3d/PrusaSlicer #11558: inconsistent ironing issue

MihRei, 2023-10-30:
### Description of the bug

so I think this is a bug. The entire top surface should had the  ironing applied but there were some lines that were missing the ironing, the printer didn't done them and I don't know why. The printer and the file had the imput shaper activated.

### Project file & How to reproduce

[Uploading ironing issue.zip…]()

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.7.0-alpha1+win64-202310191602_signed

### Operating system

windows 11

### Printer model

prusa MINI + fw 5.1.0 alpha 2

rtyr, 2023-11-02:
I don't see any issue in the g-code. The results of ironing may not be 100% reliable in general, especially on large surfaces, with different materials, etc. Some of the downsides are documented on our knowledge base. You may experiment with ironing flow and speed.

TinkerNinja, 2024-01-03:
Have the same issue - MK4 latest firmware (5.1.2), latest version of the slicer (2.7.1+MacOS-x64)

First two examples are with Prusament PETG Recycled (newly opened roll)
[image]
[image]

Rest is with Prusament PLA (Vanilla White and Galaxy Black)
[image]
[image]
[image]

Happy to provide any additional information or code if needed...

Guess ironing needs some ironing out ;)

Thanks for looking into this

closed 2026-09-29 by prusaslicer-bot[bot]