prusa3d/PrusaSlicer #11560: No choice of adding a stop or custom G-code if using multiple extruders

MaciX1337, 2023-10-31:
### Description of the bug

No choice of adding a stop or custom G-code if using multiple extruders.
I made a preset Orginal prusa I3 MK3S + MK3S +0.6 nozzle and on that preset I can not add a pause
Only choice if pressing "+" button is :"Change extruder"

Note: I only added multiple extruders so I could print different materials on top of each other easily

### Project file & How to reproduce

I made a preset Orginal prusa I3 MK3S + MK3S +0.6 nozzle and on that preset I can not add a pause

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.5.0+win64

### Operating system

Windows 10

### Printer model

MK3S

rtyr, 2023-10-31:
You have to use right mouse button.

rtyr, 2023-11-01:
I am closing the issue. We can reopen it if needed.

closed 2023-11-01 by rtyr