prusa3d/PrusaSlicer #11562: Crash when changing object height and slicing

idaadi, 2023-10-31:
### Description of the bug

Prusaslicer crashes each time when slicing while trying to correct z height of one object and its instances.

### Project file & How to reproduce

[HaloweenToys.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/13220911/HaloweenToys.3mf.zip)
Load, move all 5 instances of toyghost object to  z height to 1.98 and slice , slicer will crash.

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.0-alpha1

### Operating system

Mac OS Sonoma

### Printer model

MK4

lukasmatena, 2023-11-01:
Reproduced, thanks. We will fix it.
SPE-2005

FidelCapo, 2023-11-07:
This issue should be fixed and the fix will be part of the next release.
Closing.

closed 2023-11-07 by FidelCapo