prusa3d/PrusaSlicer #11564: perimeters modifier weakens model

nergen, 2023-11-01:
### Description of the bug

After adding box modifier and changing perimeters from 2 to 3 it basically splits model into 3 different parts in terms of 3d printing which significantly reduce it's strength.

I assume ideally the minimum number of perimeters should be continuous along all the model perimeter.
Also infill should'n't be split if it's possible.
There is no need to create an additional perimeter inside the model and split it into 3 parts.

### Project file & How to reproduce

[Middle Roller Mount.zip](https://github.com/prusa3d/PrusaSlicer/files/13223263/Middle.Roller.Mount.zip)
without modifier
[image]
After adding box modifier
[image]
[image]

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.6.1+win64 PrusaSlicer-2.6.1+win64-202309060711

### Operating system

Windows 10 (build 19045), 64-bit edition

### Printer model

Prusa MK3S

neophyl, 2023-11-01:
Its the way modifiers work and always have.  Its one of their weaknesses.  They split the object into different areas to slice depending on your settings.  This has been reported numerous times before.  

For example https://github.com/prusa3d/PrusaSlicer/issues/8121 or 
https://github.com/SoftFever/OrcaSlicer/issues/1758
There are many more if you search both open and closed issues.

Jan-Soustruznik, 2023-11-01:
The way the modifier volumes work with current implementation is they split the object into pieces, which are sliced independently. It may be improved in the future.

I am closing this as a duplicate.

Thank you, have a nice day

closed 2023-11-01 by Jan-Soustruznik